psas/Launch-12 #61: Verify Telemetrum operation

andrewgreenberg, 2015-05-01:
We had real communication problems with the Telemetrum at t Launch 11, and it never got GPS lock. This needs to be investigated, in the context of bugs #52 #60 #17.

dsigma, 2015-06-11:
TM v2.0 has been updated to the 1.6 firmware version and is correctly transmitting data to the tele dongle. This is working in my office, but may or may not work while on board the rocket.

andrewgreenberg, 2015-06-22:
OK, TM v2.0 + TeleDongle #1 now officially tested and working. Would still like to update the firmware on those, however, so now we need connectors.

LateralGs, 2015-07-05:
Connectors will be soldered up on Tuesday once the NSR is ready.  I will also be bringing a netbook to the launch running linux so we could also use that to run the telemetrum software.

dsigma, 2015-07-05:
Rad! I would much prefer if a few people/computers were able to run the TM software. I tend to get busy and last time I was unable to do it.. It can be downloaded here: http://altusmetrum.org/AltOS/

dsigma, 2015-07-13:
Tested TM with latest firmware, new igniters and packet radio comms. All worked as expected.

closed 2015-07-13 by dsigma