psas/Launch-12 #63: Store video on the NUC as well as on the camera

vosechu, 2015-05-27:
Just in case we can't get the rocket back

philiparola, 2015-05-28:
Depending on the streaming protocol, it looks like VLC might be able to do this.

vosechu, 2015-05-29:
Hey! You weren't able to see what we did on Tuesday after you left and I realize it's not in Github anywhere. We switched to using gstreamer and ended up with a smooth, clear stream (though it's small). But we did it by bypassing the trackmaster so Kenny was going to work on fixing that a bit. 

Anyways, there's scripts on the raspicam called go_camera_go.sh (stream and store) and go_camera_go_debug.sh (stream only). There's also a script in ~/bin/seecamera.sh which pops open an opengl window with the video. 

VLC might be a good option for viewing and storing at the same time, but we couldn't get it working. The source gstreamer on the raspicam is not properly encoding an SDP resource or something like that.

aSmig, 2015-07-11:
The NUC now starts a tcpdump for all traffic it sees on the tm4k network as soon as that interface comes up.  This will soak up an emergency backup of all telemetry, video stream, etc.  Not as pretty as saving to a video file, but it will get us through this launch.

closed 2015-07-11 by aSmig