psas/Launch-12 #66: Add 100nF cap to RESET_N line on RNH

andrewgreenberg, 2015-05-27:
*Just* in case it's some weird noise on the reset line. Don't believe it, but hey, possible. Also, make sure Theo implements the JTAG pin stuff, and maybe adds a watchdog.

ThirteenFish, 2015-05-30:
When firmware projects are built for flight (`make flight` instead of just `make`) a watchdog is automatically enabled.

aSmig, 2015-06-19:
@ThirteenFish does a watchdog kick result in power-cycling everything connected to the RNH?  If so, we should probably do more run testing with make flight.

andrewgreenberg, 2015-06-19:
Yes, a watchdog reset will bring down all of the power switches, rebooting everything in flight. There's a note in the RNH "v2 issues" for putting a latch between the STM32 and the power switches for exactly this reason... but that's not going to happen for this flight, obviously.

aSmig, 2015-06-20:
Is it better to have an unresponsive RNH or a full system reboot?  I'd lean towards turning off the watchdog and using something low-tech to reset RNH only while it is on the ground and frozen.  Or if it is possible to disable the watchdog dynamically, only leave it running until launch-detect.  For extra fancy points, it would be nifty to adjust the watchdog timeout on launch-detect to be approximately the flight duration plus some buffer, then return to normal back on the ground.  "Altitude based watchdog timeout?"

andrewgreenberg, 2015-06-21:
OK, if we continue to have freeze ups, I like this idea: we keep watchdogs on when we're on the ground, but at launch detects we disable them. I'd MUCH rather have an STM32 free-up on the RNH than a reboot. That would totally suck.

ThirteenFish, 2015-06-21:
This can be done, and there is some precedent for it. At NASA they will have probes disable watchdogs during critical phases, for example the Curiosity rover disabled watchdogs during re-entry.

andrewgreenberg, 2015-06-21:
Agreed on the no-need for turning back on. This is a good patch until we figure out what the hell is going on.

aSmig, 2015-07-03:
After discussion with @LateralGs and @andrewgreenberg, an externally accessible physical reset method will be made available.  We definitely still don't want to fly with a watchdog.  But training the watchdog to be aware of launch detect may not be critical.

andrewgreenberg, 2015-07-10:
This has been brilliantly solved by Kenny and his small yappy dog sidekick, Andrew. There's now a small port on the top of the avionics module that you can stick in a small wire. pushing the wire into the port slides it through a heat shrunk tubing that then goes into a 0.1" female connector. Touching the other side of the wire to the airframe tubing will reset the RNH.

Done!

closed 2015-07-10 by andrewgreenberg