psas/Launch-12 #70: Reimage picam1

aSmig, 2015-07-14:
The SD card in picam1 stopped booting.  Jon brought in 4 good cards with a history of working well with Raspberry Pi's.  Time to cook up some fresh images.

aSmig, 2015-07-15:
Started automating the image building process because it was easier than writing complete step-by-step instructions in human.

https://github.com/psas/camera-automation

Needs a bunch of stuff that will be more obvious when it is run for the first time.

vosechu, 2015-07-15:
I also brought in some sd cards recently if Jon's cards don't work. They're in a little clear box that I think is labeled backups. There's also an SD/etc card reader in that box.

aSmig, 2015-07-15:
picam1 is freshly imaged and tested.  I went ahead and imaged picam2 because @andrewgreenberg said I wasn't allowed to sleep until it was done.  Well, maybe not that last part.  But it was a good idea to get it done anyway.  Now both picams are running on the Samsung EVO 16GB cards provided by @jonhannis.  Unfortunately #72.

closed 2015-07-15 by aSmig