psas/Launch-12 #72: picam2 camera Flat Flex Cable issues

aSmig, 2015-07-15:
This cable seems extra fiddly suddenly.  It is a little mangled where it was pinched when inserting into the payload module.  We should probably just replace it.  At present the camera is not detected.  A little while ago, the camera red light came on and stayed on whenever the pi had power.  It was working right for a few minutes in between.

So we might be able to coax it to get us through this launch, but if we really want this video, we should probably replace the cable.

andrewgreenberg, 2015-07-15:
Anybody know the Digi-Key part number for this?

aSmig, 2015-07-15:
Looks like these are the only options at DigiKey: [732-5128-ND](http://www.digikey.com/product-detail/en/686715100001/732-5128-ND/4573356) (4") and [732-5129-ND](http://www.digikey.com/product-detail/en/686715152001/732-5129-ND/4573357) (6").  Or you can request whatever length you want from [Samtec](https://www.samtec.com/technical-specifications/default.aspx?SeriesMaster=FJ).   FJ-15-R-05.00-4 would be 5".  They tend to ship samples fast enough to work with our schedule.

@jonhannis might just have one floating around that is at least in better shape than ours.

aSmig, 2015-07-17:
@jonhannis had a spare camera module with good cable.  We swapped cables, then swapped cameras and still had the same "check your camera connection" error.  Turns out the Raspberry Pi was "bleeding 5v onto the 3.3v rail" or something.  Modded up a new Pi and everything is groovy.

closed 2015-07-17 by aSmig